iot-lab/iot-lab #311: FREERTOS LORAWAN on ST B-L072Z-LRWAN1 in saclay

andreavitaletti, 2024-03-27:
Hi all

GOAL: run  lorawan on ST B-L072Z-LRWAN1 with freertos

I can compile the code available on https://github.com/Lora-net/LoRaMac-node on a saclay lorawan node specifying as platform the B-L072Z-LRWAN1

The only adjustment to the cmake command is -DTOOLCHAIN_PREFIX="/opt/gcc-arm-none-eabi-4_9-2015q1/"  and clearly -DBOARD="B-L072Z-LRWAN1"       -DSECURE_ELEMENT="SOFT_SE"         -DSECURE_ELEMENT_PRE_PROVISIONED="OFF"

The build produces and ELF ... at least judging from the HEX content of the LoRaMac-periodic-uplink-lpp

I then follow the instructions available at  https://www.iot-lab.info/legacy/tutorials/riot-ttn/index.html to upload the firmware 

`iotlab-node --flash LoRaMac-periodic-uplink-lpp -l saclay,st-lrwan1,10`

However, when I connect by `nc st-lrwan1-10 20000` I get binary characters. So something is going wrong. 

I also tried to compile the serial_echo available on https://github.com/iot-lab/openlab, the freertos specifically for M3, but again the serial is not working as expected and I get binary characters

Any idea on the possible problem? 

Thanks, Andrea

aabadie, 2024-03-27:
Hello, 

Not sure but it seems the baudrate configured by LoRaMac-Node [1], 921600, is not compatible with the serial client used by IoT-LAB, 115200. 
The openlab code base is built for iotlab-m3, which is ARM Cortex-M3 (STM32 F1), and thus is not compatible with ST B-L072Z-LRWAN1 since this boards embeds an ARM Cortex-M0+ (STM32 L0). 

Regards, 

Alex 

-----

andreavitaletti, 2024-03-27:
Hi Alex 

thanks a lot ... I should have thought about it :-) 

Anyway I changed the baudrate to 115200 as you suggested in [src/boards/B-L072Z-LRWAN1/board.c](https://github.com/Lora-net/LoRaMac-node/blob/dcbcfb329b4a343ab007bc19ac43a8dc952b3354/src/boards/B-L072Z-LRWAN1/board.c#L128) the line with 
UartConfig( &Uart2, RX_TX, 115200, UART_8_BIT, UART_1_STOP_BIT, NO_PARITY, NO_FLOW_CTRL );

An now I can properly see the output on the serial 

So I can now move to properly configure the LORA connection! 

Regards, Andrea